morse-simulator/morse #769: ROS middleware installation notes mistake

EngelOfChipolata, 2017-10-10:
On this piece of documentation : [ROS Installation Notes](http://www.openrobots.org/morse/doc/1.3/user/installation/mw/ros.html) in order to install pyYAML, rospkg and catkin for Python3 it is suggested to execute `sudo python3 setup.py install`.
According to this [SO topic](https://stackoverflow.com/questions/18199853/error-could-not-create-library-python-2-7-site-packages-xlrd-permission-den), `sudo` must not be used to install with python. Furthermore, I had errors installing the packages using `sudo` : `error: could not create 'src/catkin_pkg.egg-info': Permission denied` while everything works fine using `python3 setup.py install --user`.
The documentation may need to be fixed.

nicolaje, 2017-10-10:
Hello,

note that I find it way easier to just run:

pip3 install catkin_pkg rospkg pyyaml

Regards,

Jeremy

severin-lemaignan, 2018-01-16:
Fixed using @nicolaje  suggestion. Thanks.

commit d8ec5799 (2018-01-16): [doc] Use pip to install ROS deps. Fix #769
closed 2018-01-16 by severin-lemaignan