morse-simulator/morse #771: Change logger level of ROS middleware to INFO (or default)

CodeFinder2, 2017-10-25:
Hi,

I would suggest to change the logger level (see code [here](https://github.com/morse-simulator/morse/blob/master/src/morse/middleware/ros/abstract_ros.py#L52)) to `rospy.INFO` or simply omit that parameter to use the ROS default. This causes various outputs during the startup of MORSE which renders the logging somehow unusable (too many lines). IMHO; `rospy.DEBUG` should only be used during development (of MORSE).

Any comments on this?

Thanks!

dgerod, 2017-12-06:
It's OK for me, I think that best solution is to omit the parameter so we will do what is default for ROS. I have just checked how to manage [vervosity](http://wiki.ros.org/Verbosity%20Levels) in ROS, and it match what you are proposing.

CodeFinder2, 2017-12-06:
Thanks for your reply, done (see https://github.com/morse-simulator/morse/pull/775 ).

closed 2017-12-08 by dgerod