morse-simulator/morse #781: Python modules are removed

nicolaje, 2018-01-26:
Following the issue reported by David Battle in https://sympa.laas.fr/sympa/arc/morse-users/2018-01/msg00013.html

It seems python modules are removed from Blender objects in MORSE.

Minimal builder file to reproduce the issue:

See attached simulation environment from david on the mailing list.

Todo: 

- [ ] create a simpler minimal environment that reproduces the issue.

---
- *MORSE version:* 1.4-147-gbd21e
- *Blender version:* v.2.76.0
- *Python version:* 3.5.2.

battledj, 2018-01-26:
Currently looking at the following areas for potential issues:

builder/abstractcomponent.py:        if not 'component_config.py' in bpymorse.get_texts().keys():
builder/environment.py:        if not 'multinode_config.py' in bpymorse.get_texts().keys():
builder/bpymorse.py:        texts_before = set(get_texts())
builder/bpymorse.py:        texts_after = set(get_texts())
builder/bpymorse.py:        return (texts_before ^ texts_after).pop()
builder/bpymorse.py:def get_texts():
builder/bpymorse.py:        return bpy.data.texts
builder/bpymorse.py:    if bpy and bpy.data.texts:
builder/bpymorse.py:        return bpy.data.texts[name_or_id]

nicolaje, 2018-01-31:
Hi David @battledj ,

after a bit of digging, this is a Blender bug: when appending a Blender object from an external file (see https://github.com/morse-simulator/morse/blob/a64dcec89931304c93ac0b18cbe1ec8f211619b7/src/morse/builder/abstractcomponent.py#L650 and https://github.com/morse-simulator/morse/blob/a64dcec89931304c93ac0b18cbe1ec8f211619b7/src/morse/builder/abstractcomponent.py#L714 in morse/src/morse/builder/abstractcomponent.py), only the Python scripts attached to the object are also imported, not the modules.

I have opened an issue on the Blender bug tracker:
https://developer.blender.org/T53960

and I prefer to wait for their input before implementing a MORSE specific fix.

battledj, 2018-01-31:
That's some good detective work Jeremy. I saw no evidence of a Morse 
issue either.
I'd be very interested in the feedback when you receive it.

Cheers,

David.

nicolaje, 2018-02-01:
So, according to the Blender developpers it is not a bug, but a feature :)
https://developer.blender.org/T53960

So we'll go the MORSE specific patch way.

battledj, 2018-02-01:
That's a bit unfortunate. I can't see how that could be construed as a 
feature.