morse-simulator/morse #783: Time drift between 'MORSE time' and 'ROS time'

warp1337, 2018-02-16:
When running MORSE for some time, like 1 hour (or so) I noticed
that the timestamp of ROS images (e.g. video camera) differs from 
what the actual ROS time says.

Add a fakerobot and e.g. this snippet:

<pre>
videocamera_top = VideoCamera()
videocamera_top.properties(cam_width=640,
                           cam_height=480,
                           capturing=True,
                           Vertical_Flip=True,
                           cam_near=0.01)
videocamera_top.frequency(CAM_FREQUENCY)
videocamera_top.add_interface("ros", topic="/pepper_robot/camera/front/",
                              topic_suffix='image_raw',
                              frame_id='CameraTop_optical_frame',
                              parent_frame_id='Head',
                              pub_tf=False)
fakerobot_topcam.append(videocamera_top)
</pre>

This is a little hard to reproduce since I have vision pipeline
that copies headers from one message to another and then
computes divers transforms. However, the basis timestamp
is the once of the source image generated by MORSE --- the
one that drifts over time.

I also believe I have found the problem:

https://github.com/morse-simulator/morse/blob/master/src/morse/middleware/ros/abstract_ros.py#L112

If you replace that line with rospy.Time.now() everything seems to be fine again.
I am currently running a longer test, but it might be a good idea to reproduce this.

Cheers,
  Flo

- *MORSE version:* morse UNKNOWN-dirty  (apt version in Ubuntu 16.04)
- *Blender version:* /usr/bin/blender... Found v.2.76.0
- *Python version:* Python 3.5.2

warp1337, 2018-02-17:
Okay confirmed. Using rospy.Time.now() fixes the problem. At least for me. @severin-lemaignan @pierriko

adegroote, 2018-02-19:
Who generate /clock ? Do you use /use_sim_time ?
I check the code, and there is maybe some issue, but clock and sensors use the same source of infromation (parent.gettime()), so the information should be consistent (if you use simulated time generated by Morse).