morse-simulator/morse #786: something wrong with builder introspection ?

pierriko, 2018-02-20:
Creating any function or class in a morse builder script fails to run with:

```
Blender Game Engine Started
Python module can't be imported - object 'atrv', controller 'Python':
[ERROR][MORSE] Uncaught exception, quit Blender.
ImportError: No module named 'calling'
ImportError: No module named 'calling'
```

Take any simulation builder script and add:

```
def toto():
  pass
```

you will get this issue.

My setup is: Fedora 26, Blender 2.79 tar.gz from d.b.o, Python 3.5.3
export MORSE_SILENT_PYTHON_CHECK=1
export MORSE_POP_PYTHONLIB=/usr/lib64/python3.6

severin-lemaignan, 2018-02-20:
A bit of backtrace would be useful, if possible...

pierriko, 2018-03-01:
```python
env.create()
```
soles this issue

closed 2018-03-01 by pierriko