morse-simulator/morse #791: Can't set a 2D nav goal using Rviz (starting from the navigation tutorial with PR2)

abiantorres, 2018-03-27:
* I'm triying to follow the MORSE and ROS navegation Tutorial from the official documentation.
  * It should be noted that I am making use of the kinetic distribution of ros and Ubuntu 16.04.
* My nav.launch looks like this (I copied all the .yaml files from the tutorial folder):
```
<launch>
    <param name="robot_description" command="cat $(find navigation)/urdf/pr2.urdf"/>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />
    <node name="map_server" pkg="map_server" type="map_server" args="$(find navigation)/maps/map.yaml"/>
    <node name="amcl" pkg="amcl" type="amcl"> <remap from="/scan" to="/base_scan" /></node>
	<node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen" clear_params="true">
	    <param name="footprint_padding" value="0.01" />
	    <param name="controller_frequency" value="10.0" />
	    <param name="controller_patience" value="100.0" />
	    <param name="planner_frequency" value="2.0" />
	    <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
	    
	    <rosparam file="$(find navigation)/morse_move_base/costmap_common_params.yaml" command="load" ns="global_costmap" />
	    <rosparam file="$(find navigation)/morse_move_base/costmap_common_params.yaml" command="load" ns="local_costmap" />
	    <rosparam file="$(find navigation)/morse_move_base/local_costmap_params.yaml" command="load" />
	    <rosparam file="$(find navigation)/morse_move_base/global_costmap_params.yaml" command="load" />
	    <rosparam file="$(find navigation)/morse_move_base/dwa_planner_ros.yaml" command="load" />
	</node>
</launch>
```
* When I execute the nav.launch file, I get this permanently:

```
SUMMARY
========

CLEAR PARAMETERS
 * /move_base/

PARAMETERS
 * /move_base/DWAPlannerROS/acc_lim_th: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 1.0
 * /move_base/DWAPlannerROS/goal_distance_bias: 9.0
 * /move_base/DWAPlannerROS/holonomic_robot: False
 * /move_base/DWAPlannerROS/latch_xy_goal_tolerance: True
 * /move_base/DWAPlannerROS/max_rot_vel: 1.0
 * /move_base/DWAPlannerROS/min_rot_vel: 0.4
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.03
 * /move_base/DWAPlannerROS/path_distance_bias: 5.0
 * /move_base/DWAPlannerROS/prune_plan: True
 * /move_base/DWAPlannerROS/sim_time: 1.3
 * /move_base/DWAPlannerROS/xy_goal_tolerance: 0.3
 * /move_base/DWAPlannerROS/yaw_goal_tolerance: 0.095
 * /move_base/base_local_planner: dwa_local_planner...
 * /move_base/controller_frequency: 10.0
 * /move_base/controller_patience: 100.0
 * /move_base/footprint_padding: 0.01
 * /move_base/global_costmap/footprint: [[0.35, 0.35], [-...
 * /move_base/global_costmap/global_frame: /map
 * /move_base/global_costmap/inflation_radius: 0.55
 * /move_base/global_costmap/laser/clearing: True
 * /move_base/global_costmap/laser/data_type: LaserScan
 * /move_base/global_costmap/laser/marking: True
 * /move_base/global_costmap/laser/sensor_frame: base_laser_link
 * /move_base/global_costmap/laser/topic: base_scan
 * /move_base/global_costmap/observation_sources: laser
 * /move_base/global_costmap/obstacle_range: 1.5
 * /move_base/global_costmap/raytrace_range: 2.0
 * /move_base/global_costmap/robot_base_frame: /base_link
 * /move_base/global_costmap/static_map: True
 * /move_base/global_costmap/transform_tolerance: 10
 * /move_base/global_costmap/update_frequency: 1.0
 * /move_base/local_costmap/footprint: [[0.35, 0.35], [-...
 * /move_base/local_costmap/global_frame: /odom
 * /move_base/local_costmap/height: 4.0
 * /move_base/local_costmap/inflation_radius: 0.55
 * /move_base/local_costmap/laser/clearing: True
 * /move_base/local_costmap/laser/data_type: LaserScan
 * /move_base/local_costmap/laser/marking: True
 * /move_base/local_costmap/laser/sensor_frame: base_laser_link
 * /move_base/local_costmap/laser/topic: base_scan
 * /move_base/local_costmap/observation_sources: laser
 * /move_base/local_costmap/obstacle_range: 1.5
 * /move_base/local_costmap/publish_frequency: 2.0
 * /move_base/local_costmap/raytrace_range: 2.0
 * /move_base/local_costmap/resolution: 0.05
 * /move_base/local_costmap/robot_base_frame: /base_link
 * /move_base/local_costmap/rolling_window: True
 * /move_base/local_costmap/static_map: False
 * /move_base/local_costmap/update_frequency: 2.0
 * /move_base/local_costmap/width: 4.0
 * /move_base/planner_frequency: 2.0
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    amcl (amcl/amcl)
    map_server (map_server/map_server)
    move_base (move_base/move_base)
    robot_state_publisher (robot_state_publisher/state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [29441]
process[map_server-2]: started with pid [29442]
process[amcl-3]: started with pid [29443]
process[move_base-4]: started with pid [29444]
[ INFO] [1522188502.109285464]: Loading from pre-hydro parameter style
[ INFO] [1522188502.188671616]: Using plugin "static_layer"
[ INFO] [1522188502.212260524]: Requesting the map...
[ INFO] [1522188502.445540711]: Resizing costmap to 4000 X 4000 at 0.050000 m/pix
[ INFO] [1522188502.547323940]: Received a 4000 X 4000 map at 0.050000 m/pix
[ INFO] [1522188502.557636865]: Using plugin "obstacle_layer"
[ INFO] [1522188502.576035902]:     Subscribed to Topics: laser
[ INFO] [1522188502.650807062]: Using plugin "inflation_layer"
[ INFO] [1522188502.836329166]: Loading from pre-hydro parameter style
[ INFO] [1522188502.930272573]: Using plugin "obstacle_layer"
[ INFO] [1522188502.941120686]:     Subscribed to Topics: laser
[ INFO] [1522188503.021690107]: Using plugin "inflation_layer"
[ WARN] [1522188503.115509787]: Costmap2DROS transform timeout. Current time: 1522188503.1154, global_pose stamp: 1522188497.3827, tolerance: 0.3000
[ INFO] [1522188503.153046140]: Created local_planner dwa_local_planner/DWAPlannerROS
[ INFO] [1522188503.162527920]: Sim period is set to 0.10
[ WARN] [1522188504.115601520]: Costmap2DROS transform timeout. Current time: 1522188504.1155, global_pose stamp: 1522188498.3989, tolerance: 0.3000
[ WARN] [1522188505.115700174]: Costmap2DROS transform timeout. Current time: 1522188505.1156, global_pose stamp: 1522188499.3984, tolerance: 0.3000
[ WARN] [1522188506.615741465]: Costmap2DROS transform timeout. Current time: 1522188506.6156, global_pose stamp: 1522188500.8977, tolerance: 0.3000
[ WARN] [1522188507.615786833]: Costmap2DROS transform timeout. Current time: 1522188507.6157, global_pose stamp: 1522188501.8972, tolerance: 0.3000
[ WARN] [1522188509.115612749]: Costmap2DROS transform timeout. Current time: 1522188509.1156, global_pose stamp: 1522188503.3965, tolerance: 0.3000
[ WARN] [1522188510.115866313]: Costmap2DROS transform timeout. Current time: 1522188510.1158, global_pose stamp: 1522188504.3794, tolerance: 0.3000
[ WARN] [1522188511.615791377]: Costmap2DROS transform timeout. Current time: 1522188511.6157, global_pose stamp: 1522188505.8787, tolerance: 0.3000
[ WARN] [1522188513.115607898]: Costmap2DROS transform timeout. Current time: 1522188513.1155, global_pose stamp: 1522188507.3946, tolerance: 0.3000
^C[move_base-4] killing on exit
[amcl-3] killing on exit
[map_server-2] killing on exit
[robot_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
* When I try to set a 2D nav goal after press a point in the map, nothing happens. Actually I'm not sure which is the origin of this problem. That's why I open this issue.
I hope someone can help me, thanks in advance.

abiantorres, 2018-03-27:
* Also, when I execute rosrun rviz rviz, I get this errors:
```
[...]
TIFFFieldWithTag: Internal error, unknown tag 0xa40b.
TIFFFieldWithTag: Internal error, unknown tag 0xa40c.
TIFFFieldWithTag: Internal error, unknown tag 0xa407.
TIFFFieldWithTag: Internal error, unknown tag 0xa407.
TIFFFieldWithTag: Internal error, unknown tag 0xa420.
[ INFO] [1522191108.965338113]: Creating 1 swatches
[ INFO] [1522191115.432827204]: Setting goal: Frame:map, Position(3.234, 1.662, 0.000), Orientation(0.000, 0.000, 0.963, 0.268) = Angle: 2.599
```

MMehrez, 2019-02-13:
Here is a complete tutorial on using the navigation stack in ros (including setting the goal for move_base) the robot used as an example is summit xls 

https://www.youtube.com/playlist?list=PLK8squHT_Uzdqdcx8LOGHz_RfteadQf7S

I thought of sharing it here as it might be useful!