morse-simulator/morse #793: Broken Blender component selection

severin-lemaignan, 2018-07-24:
As of January, I have been relying on the change Severin had made in d0e7ab5c77cbfe2151a210c57c0fce4b160538eb to allow selection of a particular object as the nominal robot "parent" without specific parenting in Blender (because it nullifies all physics and dynamic constraints).

Having just pulled the latest source out of GitHub, it seems that this behaviour is no longer working?

severin-lemaignan, 2018-07-24:
Commit c356bdda621c492db8652 introduced the breaking change.

The main change introduced in by this commit is to only import the actual required Blender object:
```python
imported_objects = self.append_meshes(objects=[blender_object_name])
self.set_blender_object(imported_objects[0])
```
instead of:
```python
imported_objects = self.append_meshes()
self.set_blender_object([o for o in imported_objects if o.name == blender_object_name][0])
```

It seems to break David's code. Would be useful to have more informations: how the component is called in the builder script, and what does the Blender file containing the objects look like.

battledj, 2018-07-30:
Apologies Severin and Jeremy. I have been traveling.

I am attaching a simplified version of the WAMV model, which has all the visual detail removed and only the collision boundaries remaining. When we sort this problem out, I would like to offer the full model, including the buoyancy scripts, to anybody who wants it. I recently visited my old Department at MIT and their RobotX team is keen to use the full model with MOOS.

I did stare at Severin's changes for a while, but couldn't figure out what was broken. I should add that I haven't fixed the model yet so that it can be easily translated or rotated. Currently it falls apart without parenting, but I think we can fix that per Jeremy's idea of temporary parenting.

Note that I have currently set the target blender object with:
`GroundRobot.__init__(self, 'wamv_sim/robotswamv.blend', 'robot', blender_object_name='platform_boundary')`
  
Cheers,
David.

[wamv.gz](https://github.com/morse-simulator/morse/files/2241030/wamv.gz)