morse-simulator/morse #808: Question about inheritance of SemanticCamera sensor

dgerod, 2018-12-15:
Hi,

While checking the code I have realized that:

The runtime class of SemanticCamera is inheriting from Camera as VideoCamera while DepthCamera does it from VideoCamera. However, in the builder class of SemanticCamera its parent is VideoCamera. 

In "src/morse/sensorrs/xxx.py":
class VideoCamera(morse.sensors.camera.Camera):
class AbstractDepthCamera(VideoCamera):
class SemanticCamera(morse.sensors.camera.Camera):

In "src/morse/builder/sensors.py":
class VideoCamera(SensorCreator):
class DepthCamera(VideoCamera):
class SemanticCamera(VideoCamera):

Does it is correct or it is a mistake? If it is incorrect I could fix it.

adegroote, 2019-01-08:
It looks correct. The two trees (morse.builder and morse.sensors) are not "strictly related", and so can have different inheritance trees.

dgerod, 2019-01-09:
Thanks for the answer.

I understand that builder and runtime are two different inheritance trees. However, it is quite strange that SemanticCamera and DepthCamera in builder three inherent from same type (VideoCamera) of node but from different type (Camera and VideoCamera) in runtime. 

As this is OK according to your response, I close this issue.

closed 2019-01-09 by dgerod