morse-simulator/morse #822: package install on Ubuntu 20.04 not working

auturgy, 2020-05-25:
Fresh install on Ubuntu 20.04 not working
Steps to reproduce:
   sudo apt install morse-simulator
   morse create my_first_sim
   morse run my_first_sim

Full console dump (of a different run, but same error) is at shorturl.at/svJM6
but basically Blender quits due to uncaught exception, 
AttributeError: 'Object' object has no attribute 'hide'.  
--------------------------------------------------------------------
09:23 $ morse --version
morse 1.4

09:23 $ morse check
* Checking up your environment...
* Running on Linux. Alright.
* Found MORSE libraries in '/usr/lib/python3/dist-packages/morse/blender'. Alright.
* Trying to figure out a prefix from the script location...
* Default scene found. The prefix seems ok. Using it.
* Setting $MORSE_ROOT environment variable to default prefix [/usr/]
* Checking version of /usr/bin/blender... Found v.2.82.7
* Version 2.82.7 of Blender is untested but should work
* Found Blender in your PATH
(/usr/bin/blender, v.2.82.7).
Alright, using it.
* Checking version of Python within Blender /usr/bin/blender... Found v.3.8.2
* Blender and Morse are using Python 3.8.2. Alright.
* Your environment is correctly setup to run MORSE.

dgerod, 2020-05-26:
My first impression is that Blender API used in Morse is not compatible with your Blender version (2.82.7), MORSE is complaining about a missed attribute "hide". But, I have not checked in detail.

I use version 2.78c. So, could yo use an older Blender version? You could try with several ones, from 2.78c to 2.82.7, if you are interested in having latest Blender version.

severin-lemaignan, 2020-05-26:
The Blender game engine has been removed from Blender 2.8x, and MORSE can not run without it. The last version of Blender that should work is 2.79.

dgerod, 2020-05-26:
@severin-lemaignan, this important information about Blender/BGE version should be written in the README to avoid people to have this problem.

severin-lemaignan, 2020-05-26:
Indeed. Can you send a quick PR with this information?

jiangxkjohn, 2021-12-15:
Hi,

I met the same problem, but I change the version of the blender to 2.79b and then this exception was disappeared.

However, I meet another problem, when I run the first demo

`morse run my_first_sim`

It came out Segmentation fault

`/usr/local/bin/morse:340: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  config = configparser.SafeConfigParser()
* Adding </home/robot/my_first_sim> to PYTHONPATH and MORSE_RESOURCE_PATH.

* Using default scene in environment <my_first_sim>.

* morse 1.4

* Version 2.79.0 of Blender is untested but should work

/usr/local/bin/morse:302: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn("Blender's Python version could not be determined. "
* Blender's Python version could not be determined. Crossing fingers.

found bundled python: /home/robot/libraries/blender-2.79b/2.79/python
Read blend: /usr/local/share/morse/data/morse_default_autorun.blend
Writing: /tmp/morse_default_autorun.crash.txt
Segmentation fault (core dumped)
`
Did you meet before and how did you solve it?

Thank you.

severin-lemaignan, 2021-12-15:
Can you post the content of `/tmp/morse_default_autorun.crash.txt`?

jiangxkjohn, 2021-12-15:
Sorry, I change another machine using ubuntu 18.04 and fix all the problems, I guess it might related to ubuntu version.
But in the previous environment, I cannot recover the exception before, and it came out with another fault.

```
/usr/local/bin/morse:340: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  config = configparser.SafeConfigParser()
* Adding </home/robot/my_first_sim> to PYTHONPATH and MORSE_RESOURCE_PATH.

* Using default scene in environment <my_first_sim/>.

* morse 1.4

* Version 2.79.0 of Blender is untested but should work

/usr/local/bin/morse:302: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn("Blender's Python version could not be determined. "
* Blender's Python version could not be determined. Crossing fingers.

found bundled python: /home/robot/libraries/blender-2.79b/2.79/python
Fatal Python error: Py_Initialize: can't initialize sys standard streams
Traceback (most recent call last):
  File "/usr/lib/python3.8/io.py", line 52, in <module>
  File "/usr/lib/python3.8/abc.py", line 106
    print(f"Class: {cls.__module__}.{cls.__qualname__}", file=file)
                                                      ^
SyntaxError: invalid syntax
Aborted (core dumped)
```

But I guess it didn't write to ```/tmp/morse_default_autorun.crash.txt``` so I copy this file below for your reference.

```
# Blender 2.79 (sub 0), Commit date: 2018-03-22 14:10, Hash f4dc9f9d68b
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.logic.sensor_add(type='ALWAYS', name="MORSE_LOGIC")  # Operator
bpy.ops.logic.controller_add(type='PYTHON')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.empty_add(type='ARROWS', view_align=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.logic.sensor_add(type='ALWAYS', name="MORSE_LOGIC")  # Operator
bpy.ops.logic.controller_add(type='PYTHON')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.empty_add(type='ARROWS', view_align=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.logic.sensor_add(type='ALWAYS', name="MORSE_LOGIC")  # Operator
bpy.ops.logic.controller_add(type='PYTHON')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.empty_add(type='ARROWS', view_align=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.logic.sensor_add(type='ALWAYS', name="MORSE_LOGIC")  # Operator
bpy.ops.logic.controller_add(type='PYTHON')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.mesh.primitive_cube_add(view_align=False, enter_editmode=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator
bpy.ops.material.new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.empty_add(type='ARROWS', view_align=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.game_property_new()  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.object.select_all(action='DESELECT')  # Operator
bpy.ops.mesh.primitive_cube_add(view_align=False, enter_editmode=False, location=(0, 0, 0), layers=(True, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False))  # Operator

# backtrace
/home/robot/libraries/blender-2.79b/blender(BLI_system_backtrace+0x20) [0x1a6c700]
/home/robot/libraries/blender-2.79b/blender() [0x1078395]
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f51a0416210]
```