morse-simulator/morse #824: Blender Version check reults in Index Error

Crazy-Ginger, 2020-06-22:
Running:
morse run [simulation]
results in a IndexError

Morse version was built in arch linux from <https://aur.archlinux.org/packages/morse-simulator-git> using yay

The issue is in: /bin/morse.in line 93
```python
version = line.split()[1] + '.' + line.split()[3][:-1]
```
line.split() is only 2 elements thus causing an IndexError

---
- *MORSE version:* 1.4-180-g9deb-dirty
- *Blender version:* 2.83.0 [from `blender -v`] 
- *Python version:* 3.8.3 [from `python --version`]

cplaursen, 2020-06-25:
From #822 
> The Blender game engine has been removed from Blender 2.8x, and MORSE can not run without it. The last version of Blender that should work is 2.79.

Crazy-Ginger, 2020-06-25:
Ah thank you

closed 2020-06-25 by Crazy-Ginger